blakadder/templates #1: Site images

blakadder, 2019-02-26:
[image]

blakadder, 2019-02-26:
[image]

blakadder, 2019-03-16:
[image]

melikyuksel, 2019-03-22:
Some images on the Template website seem to link directly to the submitted image link. Will these images be moved to a more permanent location in case of links being broken?

blakadder, 2019-03-23:
No. In the template form it is explicitly stated to host the images somewhere else just because of that

melikyuksel, 2019-03-25:
Okay, here's one then. [image]

closed 2019-04-08 by blakadder