blakadder/templates #4: Please do not remove templates from the wiki

kueblc, 2019-03-02:
Users do not need to fork your repository or use a google form to edit the official wiki. All of the information is in one place and should remain accessible as such.

It's fine that you want to provide another option for people to organize templates but it's not okay to remove from the wiki.

It's useful to have all the device templates on one page, even if they are from different countries, as many of the device templates apply to devices regardless of country.

It's also difficult to get info at a glance when info is split up between multiple pages.

I kindly request that you refrain from deleting and reverting wiki pages from the official Tasmota wiki. As an alternative I offer that you keep a link in the sidebar, but clearly mark that it your project.

blakadder, 2019-03-02:
I'm the wiki editor for Tasmota and it was a decision of the entire development team to use the github pages version since it offers more powerful features. So please stop editing the sidebar despite the request in Contribution article not to do so

kueblc, 2019-03-02:
Could you elaborate on "more powerful features"?

Jason2866, 2019-03-02:
@kueblc 
Please DO NOT revert the changes from @blakadder in Tasmota wiki
He is the official wiki maintainer!
Please respect Theo Arendst project
Thank you!

kueblc, 2019-03-02:
@Jason2866 No disrespect intended! The fact that the community relies on this information and that this recent unannounced change makes it difficult to access this information *is* an issue.

Please consider the user. Making it more difficult to contribute and find information is not user friendly.

Jason2866, 2019-03-02:
@kueblc You are not the person to decide! Theo himself has @blakadder anounced to to tje wiki
If you dont respect this you will get banned from Tasmota github

kueblc, 2019-03-02:
Of course the project owner decides what happens to their project. I am not debating that.

Please address the issues.

blakadder, 2019-03-02:
> @Jason2866 No disrespect intended! The fact that the community relies on this information and that this recent unannounced change makes it difficult to access this information _is_ an issue.
> 
> Please consider the user. Making it more difficult to contribute and find information is not user friendly.

How is the new site making it more difficult opposed to the huge table in the wiki???

kueblc, 2019-03-02:
As an aside @Jason2866, I'm finding it unnerving to see you continually arguing with my comments across github. Please do respect my space. I have no issue with you, Theo, Tasmota, or @blakadder.

I have no problem "respecting" the Tasmota wiki, this unannounced change did not appear legitimant so I'm sure you can understand why I would want to protect users.

As far as the difficultly in using this system, as I outlined in the first post
- Users need to fork or use a Google form in order to add devices, rather than being able to edit directly
- You can only view one country at a time, when devices might be relevant to many regions
- You need to navigate multiple pages in order to reach the content
- The "huge table" allows you to quickly scroll to find the picture of the device you're looking for

kueblc, 2019-03-02:
If you are set on pursuing this system may I suggest adding a section that mirrors the original content "huge table", maybe a "view all" option.

blakadder, 2019-03-02:
> I have no problem "respecting" the Tasmota wiki, this unannounced change did not appear legitimant so I'm sure you can understand why I would want to protect users.

Noone complained when I unannouncedly added the template page in the first place
> 
> As far as the difficultly in using this system, as I outlined in the first post
> 
> * Users need to fork or use a Google form in order to add devices, rather than being able to edit directly

If someone is set on editing an existing template he'll manage the hurdle. If it was open for editing to anyone we get situations like this

> * You can only view one country at a time, when devices might be relevant to many regions

Is that a joke?

> * You need to navigate multiple pages in order to reach the content

Opposed to scrolling to reach the content

> * The "huge table" allows you to quickly scroll to find the picture of the device you're looking for

If you have a device you already know which category it belongs to and what standard your country uses

kueblc, 2019-03-02:
As for the editing issue, can you create a version that does not rely on Google forms? I do not think I am alone, especially in a community focus on going cloudless, that I prefer not to use Google services.

blakadder, 2019-03-02:
> If you are set on pursuing this system may I suggest adding a section that mirrors the original content "huge table", maybe a "view all" option.

wow? how come i didnt think of that?!?!

blakadder, 2019-03-02:
You're welcome to code me a form that automatically converts to required .markdown and host it somewhere for me

kueblc, 2019-03-02:
No, I'm not joking, and I'm insulted that you are not taking feedback seriously. These changes impact usability. A powerstrip from EU might use the same template as one from the US. I have to check each region's page in order to find a device similar to the one I'm looking for.

blakadder, 2019-03-02:
> No, I'm not joking, and I'm insulted that you are not taking feedback seriously. These changes impact usability. A powerstrip from EU might use the same template as one from the US. I have to check each region's page in order to find a device similar to the one I'm looking for.

have you thought about looking in the plugs section instead?

kueblc, 2019-03-02:
That section did not exist when the wiki page was replaced. I have not been following changes to your project. Looking now, it does seem that most of the issues I had have already been addressed so I will close this issue.

I have to say that I've repeatedly experienced abrasive communication from the Tasmota group, unlike my experience with most other open source projects. I'd ask that you consider assuming good faith and approach users kindly to reach mutual understanding. I do not feel welcomed to the community and regret contributing to the project as I've been treated with nothing but disrespect since then.

closed 2019-03-02 by kueblc

blakadder, 2019-03-02:
> That section did not exist when the wiki page was replaced. I have not been following changes to your project. Looking now, it does seem that most of the issues I had have already been addressed so [...]

You do realise everything done on the wiki is volunteer work and we cannot implement features on your request.

kueblc, 2019-03-02:
Yes, I do understand how wikis and open source works. I'm not asking anyone to drop everything and do something for me, I was expressing that the change created problems for me and likely others.

I contributed support for the SM16716, enabling use of Tasmota on a wide range of RGB light bulbs. Regardless, whether or not I contributed, users should be treated with respect. I did not know you as a contributor either. I apologize for the miscommunication on my part, please do consider that you could be more receptive to feedback as well.

jziolkowski, 2019-03-02:
@kueblc what you are seeing with the split between wiki and github pages done by @blakadder is a part of possible transition to a completely different platform. So hold your horses. You don't know of all the decision making behind the scenes, all the pros and cons being discussed, etc. Mind the fact, that people in the know have a much broader perspective over what's currently happening. Your reverts - while no one doubts good intentions - generate additional, unnecessary work for us.

also 

> community focus on going cloudless

you know github is a cloud-based website, right? Cloudless is having a local hosted copy, which you're free to fork, and add to it in whatever way you see fit.