blakadder/templates #10: Digoo DG-SP01 not working

killah1703, 2019-03-18:
[That ](https://github.com/blakadder/templates/blob/master/_templates/digoo-dg-sp01.markdown)template is not configured correctly. Unfortunately, the device becomes extremely unstable after activating the template. A reset is almost impossible. ***I do not recommend it.***

What or who is the origin of [this ](https://github.com/blakadder/templates/blob/master/_templates/digoo-dg-sp01.markdown)configuration?
Can I resolve the mappings without opening the device? There are no screws. probably glued.

Tested with:
Tasmota Version: 6.5.0 (sonoff) & 6.4.1.21 (sonoff)
Core / SDK Version: 2_5_0 / 3.0.0-dev (c0f7b44)

https://github.com/blakadder/templates/blob/master/_templates/digoo-dg-sp01.markdown

blakadder, 2019-03-18:
are all the features working besides the instability of the device itself?

unstable means what exactly? is the device resetting or disconnecting from mqtt and/or wifi?

killah1703, 2019-03-18:
If I use anything in the web UI, e.g. On / Off or just the Settings or Console button, the device will hang and cause timeouts or freezes completely. The hardware button only reacts occasionally.
For me it was a gamble to reset the device in this state. **Do not use it.** :)

There are 2 relays in the template, but the device has only 1 afaik.

I will try to create a new template and will give feedback.

blakadder, 2019-03-19:
I have added a warning to the template until it gets changed. 
Thank you for your help!

closed 2019-03-19 by blakadder

ezcell, 2019-11-19:
Hey, have you managed to figure out the right template config for this plug?