blakadder/templates #23: I need help with templates

demian85, 2019-04-23:
Hi,

I'm familiar with Tasmota and I have already flashed many Sonoff devices at home, but I just bought a pack of [2 light bulbs](https://www.amazon.com/dp/B07DLSNNDS) and all the details I found in the Tasmota Wiki is the section "device templates". I'm not sure what is the purpose of that.

Questions:

- Can you please explain how to use a device template?
- Do you have any idea which steps to follow in order to flash these bulbs with Tasmota? I cannot even open them.

Thanks!

demian85, 2019-04-23:
Nevermind, someone just shared this link in the support chat: https://github.com/arendst/Sonoff-Tasmota/wiki/Templates

I could not find it before!

Anyway, I still don't know how to flash the bulb, any help is appreciated. Thanks!

blakadder, 2019-04-24:
by using https://github.com/ct-Open-Source/tuya-convert unless you want to disassemble the bulb

closed 2019-05-01 by blakadder