blakadder/templates #28: Not working on 6.5.0

whatsupskip, 2019-05-11:
I have read elsewhere that the Brilliant unit and the Genio have different configurations. 

Toggling the switch doesn't work, nor does using the switch on the device itself.

blakadder, 2019-05-11:
I wouldn't know, these are all submitted templates which means they worked for someone. Manufacturers often change the interior design without changing the shell or the product name

whatsupskip, 2019-05-12:
I have since found that the following template works: {"NAME":"Brilliant","GPIO":[0,0,0,0,0,21,0,0,0,52,90,0,0],"FLAG":0,"BASE":18}. Can this be added to the notes?

blakadder, 2019-05-12:
it can but i need to know which the device link first

whatsupskip, 2019-05-17:
The Brilliant WiFi switch with USB charging from Bunnings.

closed 2019-05-17 by blakadder