blakadder/templates #45: NEO Coolcam 16A (outdated)

localhost61, 2019-06-13:
The Coolcam 16A device I just received from Aliexpress holds a BL0937 instead of a HLW8012.
- As Theo suggested in his response to the [issue](https://github.com/arendst/Sonoff-Tasmota/issues/5945) I had to get consistent Energy values , in [your template](https://blakadder.github.io/templates/neocoolcam_16a.html) that I used, GPIO4=131 should be replaced with GPIO4=134.

- In the same page, it looks like the szneo website is outdated, there is no 16A plug in the catalog, only 10A.

blakadder, 2019-06-13:
that doesn't mean the current template is not operational for all devices

localhost61, 2019-06-13:
But it deserves a second template or a side note for new comers. Doesn't it?
It just happens that your template doesn't fully match my device, I sent pictures, I've nothing to compare with previous version, if any.

blakadder, 2019-06-13:
It seems you have missed the quite large Add new template button on the main page of the site

meingraham, 2019-06-13:
@localhost61 Xavier, an alternative is to edit the existing template entry (fork it and PR) and add the second template with the respective "Here's template for this class of NEO Coolcam and second template for this class with different power monitoring chip". I had this situation with the Kesen KS-604S. This way there aren't two template entries for the same bulb "model" just the variant templates within that template entry.

closed 2019-06-15 by blakadder