blakadder/templates #51: Unable to add new template

macadamer, 2019-06-23:
Tried to add a Maxcio UK power plug template for model W-UK007 (not the power-monitoring variant already in the Templates list), but the form queried if I was sure the template was correct. Presumably because of the already-present UK007S. Anyhoo, mine came from Amazon, it was Tuya-convertible, and seems to work fine.

Anyway, here it is, hope it's OK.

 {"NAME":"Maxcio","GPIO":[0,17,0,0,0,0,255,255,0,56,21,0,0],"FLAG":0,"BASE":18}

blakadder, 2019-06-24:
The template is regex checked so something is off either with the template or the regex. Since this is the first time i'm inclined to believe it was the template

macadamer, 2019-06-24:
Still says "Are you you sure this is the output of Tasmota template command?" Yes, I am. I used the template on another plug and that works fine too. Hmm.

Screenshot from "Configure Template":

[image]

Screenshot fron Console:

[image]

blakadder, 2019-06-24:
Don't know what to tell you, I ran the template from the op through the google form and it submitted without errors

macadamer, 2019-06-24:
Would you mind therefore adding the template to the repository for me please? Could use same pic as the W-UK007S already present. I purchased these plugs from Amazon UK, but looks like they only have the UK007S model here now.

closed 2019-06-24 by macadamer