blakadder/templates #69: dimmer slider and warmth for white slider only

dasb00ter, 2019-07-23:
I just got the 3 pack for this bulb. With sonoff 6.5.0 Im getting a slider for warm white to cold white and a brightness/dimmer slider but no color slider. Confirmed rgb with color codes that work

blakadder, 2019-07-23:
don't know which bulb but either the template is wrong or you mixed up the bulbs

dasb00ter, 2019-07-23:
I just looked at the packaging and confirmed it's the correct upc code. Model # on the bulb 37726. Thanks for confirming I should see a color slider. Do u have 3 sliders for brightness, color and white warmth (2000-5000k)? Thanks for putting out the template it got me going even if it seems a bit wonky for me. I guess the only other thing worth mentioning is the only way I can turn them off in home assistant after setting them up with mqtt is to turn the brightness to zero. In the sonoff console power commands work fine though

closed 2019-10-23 by blakadder