blakadder/templates #102: Device mis-classifications?

meingraham, 2019-09-14:
What characteristic in the template makes the [Shelly RGBW2](https://blakadder.github.io/templates/shelly_RGBW2.html) Miscellaneous and the [MagicHome](https://blakadder.github.io/templates/magichome_ZJ-WFMN-A_RGBW.html) not? What needs to be changed in the Shelly so that it's "together" with the other similar devices?

Should the Slampher by classified with the bulbs?

Mike

P.S. You've been a stranger lately. YOu must have a heavy work project load. Drop by Discord sometime :wink:

blakadder, 2019-09-14:
The people submitting the device!

No

I am in Discord all the time

meingraham, 2019-09-14:
I tried to look at the Shelly entry to see where to change it. I am happy to re-classify it but didn't see where that happens.

Mike

P.S. Discord - Cool. You're just quieter than before the summer holidays.

blakadder, 2019-09-15:
No worries, I changed it as soon as I read this issue

closed 2019-09-15 by blakadder