blakadder/templates #131: Teckin SB50 missing Cold White GPIO

nebula-it, 2019-10-14:
I got the GPIO connections from https://blakadder.github.io/templates/teckin-sb50.html for my Teckin SB50, but noticed that it was missing the control for white light temperature. So I opened one up and found that GPIO5 is connected to the cold white LED's. I'm not sure how tasmota handles GPIO assignment so here is the complete list of GPIO's for this bulb. Please update the template accordingly:

| GPIO  | LED connected |
| ------------- | ------------- |
| 4 |  Red |
| 5  | Cold White |
| 12 | Green |
| 13 | Warm White |
| 14 | Blue |

Let me know if you need any additional info on this. Cheers

digiblur, 2019-10-17:
There are two hardware revisions of the SB50, one without 2 white channels.

nebula-it, 2019-10-18:
Hey @digiblur ,
First of all, I wanted to say thank you for the awesome work that you do. It was your post on Reddit about updated tuya-convert that got me working with these bulbs and I was able to flash these.

Secondly, so for the bulbs with two white channels how do those control the temperature for white light? Or do they not come with that option at all?

digiblur, 2019-10-18:
5 channel bulbs should work by testing with the slider on the Tasmota GUI. Or try these commands on the console

Color 255,0,0,0,0 
Color 0,255,0,0,0 

And so forth moving the 255 down the line. Be careful not to assign 255 to too many channels at one time as you can overdrive the bulb power supply.

closed 2019-11-21 by blakadder