blakadder/templates #156: Lohas 9W RGBW

Vendo232, 2019-10-28:
used this template and following Light setup in HA and the bulb is not showing up as COLOR light bulb. Please help. Using Tasmota 6.7.1

{"NAME":"Lohas RGBW","GPIO":[0,0,0,0,0,0,0,0,0,143,0,144,0],"FLAG":0,"BASE":18}

  - platform: mqtt
    name: "Lohas1"
    command_topic: "cmnd/Lohas1/POWER"
    state_topic: "tele/Lohas1/STATE"
    state_value_template: "{{value_json.POWER}}"
    availability_topic: "tele/Lohas1/LWT"
    brightness_command_topic: "cmnd/Lohas1/Dimmer"
    brightness_state_topic: "tele/Lohas1/STATE"
    brightness_scale: 100
    on_command_type: "brightness"
    brightness_value_template: "{{value_json.Dimmer}}"
    white_value_state_topic: "tele/Lohas1/STATE"
    white_value_command_topic: "cmnd/Lohas1/white"
    white_value_scale: 100
    white_value_template: "{{ value_json.Channel[3] }}"
    rgb_command_topic: "cmnd/Lohas1/Color2"
    rgb_state_topic: "tele/Lohas1/STATE"
    rgb_value_template: "{{value_json.Color.split(',')[0:3]|join(',')}}"
    effect_command_topic: "cmnd/Lohas1/Scheme"
    effect_state_topic: "tele/Lohas1/STATE"
    effect_value_template: "{{value_json.Scheme}}"
    effect_list:
      - 0
      - 1
      - 2
      - 3
      - 4
    payload_on: "ON"
    payload_off: "OFF"
    payload_available: "Online"
    payload_not_available: "Offline"
    qos: 1
    retain: false

blakadder, 2019-10-28:
https://community.home-assistant.io/ or https://discord.gg/c5DvZ4e

closed 2019-10-28 by blakadder

Vendo232, 2019-10-28:
Your template is not working, sending me default links won`t help. Searched the forum and discord and haven`t found single person who would claim this bulb works on latest Tasmota.

blakadder, 2019-10-28:
Apparently it worked for the person that submitted the template. Either that or that same person wants to sell a lot of bulbs so they lied that it works with Tasmota