blakadder/templates #162: Conventions for GPIO mapping?

pfeerick, 2019-11-03:
I was about to submit a PR for https://github.com/blakadder/templates/blob/master/_templates/medion_life_S85225 with a different GPIO mapping, and then realised that GPIO13 has been mapped to `LedLinki1(158)` instead of the `Led1i(56)` I used. Has any standard or convention been set thus far for this sort of setting, or is this strictly personal preference territory? I couldn't see anything the Tasmota wiki/template site, so perhaps this is still something to think about?

ie. I set it like this
`{"NAME":"Medion S85225","GPIO":[0,0,0,17,134,132,0,0,131,56,21,0,0],"FLAG":0,"BASE":18}`
as opposed to 
`{"NAME":"Medion","GPIO":[0,0,0,17,134,132,0,0,131,158,21,0,0],"FLAG":0,"BASE":52}`

blakadder, 2019-11-03:
see https://github.com/arendst/Tasmota/wiki/Status-LED for when is what used

pfeerick, 2019-11-04:
Thank you for the link - I do know what the settings are. 

My question was: 
> Has any standard or convention been set thus far for this sort of setting, or is this strictly personal preference territory?

Now that this excellent template system exists (nice work btw!), in order to standardise the mapping so people know what to expect when using the templates, what are the conventions for GPIO mapping? i.e. Single LED == use LED1(i), Dual LED = use LED1(i) and then LedLink1(i), etc, etc.

blakadder, 2019-11-04:
There are no "conventions" its just a matter of how you want to use that single led. If there's two then logic dictates one is meant for power and one for wifi status. In any case, its a template, who doesn't like it can change it so you just submit a template tailored to your needs

closed 2019-11-21 by blakadder