blakadder/templates #197: Template for AOFO 4AC+4USB not working

realex-me, 2019-11-23:
Hey blakaddr,
A few days ago I bought the new Smart Power Strip „AOFO 4AC+4USB “ on Amazon. Yesterday I flashed it with Sonoff-Tasmota 6.7.1 by Theo Arends and configured it with your template but as result i got only one Toggle Button which toggles the state of Plug 1 (see image below). Are there any additional configurations I have to do?

[image]

blakadder, 2019-11-23:
There are a few power strips named that. Maybe you have a different version or your template isn't applied correctly

blakadder, 2019-12-03:
Another version of the template added

closed 2019-12-03 by blakadder