blakadder/templates #216: H801 template does not have white channel

robin13, 2019-12-02:
When applying the h801 [template](https://blakadder.github.io/templates/h801.html), the web interface (and reported auto-config) does not seem to contain a channel for switching the Warm-White LEDs:

[image]

robin13, 2019-12-02:
Tasmota 6.7.1 BTW and the white light is connected to W1

blakadder, 2019-12-03:
Thanks. I'll check and update once I get access to my H801

robin13, 2019-12-03:
Having updated to 7.1.1 I now see:

[image]

So actually now all of the elements are visible, BUT when I change the warm/cool slider, the coloured LEDs go out, and when I change the colour selector slider the white LEDs go out.. it seems as if each command automatically turns off the "other" ... so I'm not sure if this is a template issue, or a limitation of tasmota?

blakadder, 2019-12-03:
you can control either rgb or white channel with sliders

closed 2019-12-03 by blakadder