blakadder/templates #219: To  @meingraham

frablaser, 2019-12-03:
Hi Mr @meingraham

In the templates I saw the DS01 US dimmer, the new release of this dimmer is no more what is shown in the picture. The hardware really changed, it is not possible to soldering the wires directly to the ESP microcontroler anymore. It is to tiny and the PCB donuts are became very tiny also (less than 1.5 mm) with a big welding nose it could cause some big damages.

In the board, there is some holes and plates where you can soldering your wires, but it is near the double of a hair size may be.
I tried to flash it, but it seems the the GPO0 isn't at the same place (put it to the GND don't affect the ESP boot) and I can't figure out how to set the ESP to be flashed... I tried with or without the flat wires plugged.  

Pictures:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

meingraham, 2019-12-03:
Did you try Tuya-Convert on this device?

This may be a case where the RST pin on the secondary MCU is keeping the ESP chip from going into boot mode. This has been reported for other Tuya devices with a secondary MCU. 

https://github.com/arendst/Tasmota/wiki/MCU-Based-Tuya-Dimmers-and-Switches#flashing---preparation

I don't have any other suggestions.

blakadder, 2019-12-03:
Definitely not a site issue.

closed 2019-12-03 by blakadder

frablaser, 2019-12-03:
@blakadder Yes not a site issue at all...  But No other way for me to contact Mr Ingraham... 😌

@meingraham 
I converted a lot of devices, but this one resist...

I will try to figure out the RST pin on the secondary MCU. Do you think I could remove the soldering on the RST on the master MCU?

meingraham, 2019-12-03:
> No other way for me to contact Mr Ingraham

Tasmota Discord - https://discord.gg/Ks2Kzd4