blakadder/templates #240: Not able to change white colors

ashoktvm, 2019-12-10:
In the template provided, we are not ab;e to change the white. (To cool white or warm white). can you add that option as well

blakadder, 2019-12-10:
i honestly don't know which template you're talking about

grumpy-dude, 2019-12-16:
This template is actually for the Merkury MI-BW9 **_0_** 2-999W, which is a dimmable, warm-white only bulb.

The template for the Merkury MI-BW9 **_4_** 2-999W (an updated version of the BW902?), which is also a dimmable, warm-white only bulb, is as follows:
{"NAME":"MI-BW942-999W","GPIO":[0,0,0,0,37,0,0,0,0,52,0,0,0],"FLAG":0,"BASE":18}

Note that both of these bulbs are warm-white only, and are _not_ color-temperature tunable.

blakadder, 2019-12-17:
I'm confused... What's the exact issue here?

closed 2020-01-04 by blakadder