blakadder/templates #256: More verbose template

mcspr, 2019-12-25:
Hey,

I've been a frequent visitor of the templates site since a lot of ESPurna users at https://gitter.im/tinkerman-cat/espurna or https://github.com/xoseperez/espruna/issues (and sometimes myself) reference some existing device that is already documented. While I do know how to parse Tasmota's template header and manually offset lines in vim or something like that, do you think it would be a useful addition to have GPIO pin and magic numbers as a hover hint or maybe some optional extended view (as a "verbose" button) to explain each number line-by-line? Enumeration as string values could be used as-is, GPIO numbers <=> array index should be pretty straight forward.

And I do see that the existing template uses hardcoded values to tell the device type (power plug, light etc.)

---
I was also thinking about having a small webpage and reference people to that to parse templates with the current tasmota_template.h, but perhaps that functionality would also be useful for other projects like ESPHome or ESPEasy

blakadder, 2019-12-25:
I have no plans to support different firmwares. Reference to the GPIO numbering is found in https://tasmota.github.io/docs/#/Components. 
I will see about having some sort of decoding of the template but its not on any immediate todo list since the current functionality is sufficient to use with Tasmota

mcspr, 2019-12-25:
Oh, thanks for the link, I had no idea about that part of documentation.

Just to follow my own idea, I came up with a little cpp program that can be compiled directly in the Tasmota repo root:
https://gist.github.com/mcspr/eb68ada9df1e4e08ac3d2a129ac7c680
For example, https://templates.blakadder.com/shelly_1PM.html
```console
$ pwd
/home/max/Documents/PlatformIO/Projects/Tasmota
$ g++ -o show_template show_template.cpp
$ ./show_template '{"NAME":"Shelly 1PM","GPIO":[56,0,0,0,82,134,0,0,0,0,0,21,0],"FLAG":2,"BASE":18}'
Shelly 1PM (based on Generic)
ADC0 Temperature
GPIO 0 -> Led1i
GPIO 4 -> Switch1n
GPIO 5 -> BL0937 CF
GPIO15 -> Relay1
```
Which seems to be correct interpretation
(and I do see the same pin mapping in the template used by espurna... but I am not sure that I translate BASE correctly)

blakadder, 2019-12-25:
[image]
Merry Christmas! Spread the word!

closed 2019-12-25 by blakadder