blakadder/templates #265: [Site issue]

majorsl, 2019-12-28:
Unsure on how to edit an existing template info or if I should.  I can create a new template if needed.

Wanted to note that this bulb: https://templates.blakadder.com/fcmila_7w.html

Is the exact same I bought on eBay.com (US) in the 9W version (still E27).  This template works perfectly for it.

The box has checkmarks for the factory to tick 7W, 9W, or 12W. I can only confirm the 9W is the same.

ascillato, 2019-12-28:
Ok. So, what is the information you want to add?

majorsl, 2019-12-28:
Perhaps just change the title to Fcmila 7W/9W would be good enough.

ascillato, 2019-12-28:
Yes, I think you are right. There are other templates of Fcmila of different power but they have also different gpio templates.

In the template, you can click on EDIT ON GITHUB link that is below the gpio table, then click on the pencil (up right of the page) and change the title key. Then you click on propose file (green button on the bottom) and then on make Pull Request.

Thanks

majorsl, 2019-12-28:
Done. Thanks!

closed 2019-12-30 by blakadder