blakadder/templates #268: wrong GPIO

bosshugo, 2019-12-29:
Hi,

I tried {"NAME":"MI-BW942-999W","GPIO":[0,0,0,0,37,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":18}
but it doesn't work. I can't turn it on/off and change brightness.

After test, it should be
{"NAME":"MI-BW902-999W","GPIO":[0,0,0,0,0,37,0,0,0,0,0,0,0],"FLAG":0,"BASE":18}
GPIO5 is PWM1

Edward

blakadder, 2019-12-29:
is the manufacturer id exactly the same?

bosshugo, 2019-12-29:
Yes, please see attached photo. I bought two kind of this bulb from
Walmart, the RGB one's template is correct. Only this White one, I can't
turn it on. So I tried others and found if I change GPIO to 5 it works.

Edward
[image: IMG_20191229_112905.jpg]
[image: IMG_20191229_112924.jpg]

blakadder, 2019-12-30:
the template is updated

closed 2019-12-30 by blakadder