blakadder/templates #285: Treatlife DS02S - Possible Template Issue.

Foppotee, 2020-01-02:
Hi. If this is the wrong place to post for this please redirect me.

I found an issue with using the Treatlife DS02S switches. I had two switches installed in two different rooms. When turning one of the switches on it would cause the lights in both rooms to come on. I am new to Tasmota and would love to assist troubleshooting this issue.

Troubleshooting so far:
- Replaced both switches one at a time.
- Moved one switch to a different wireless network.

I noticed the effects were different depending on what the dimmer level was set to. I uninstalled the setup as those rooms are my kids rooms and they need their lights (without risk of electrocuting themselves.) I am going to try to rebuild the circuit in my garage to attempt to replicate the issue.

closed 2020-01-02 by blakadder

Foppotee, 2020-01-02:
@blakadder Would you please point me to a resource for discussing the issue above?

ascillato, 2020-01-02:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4).

Thanks.

blakadder, 2020-01-02:
You see, when you go and open an issue there's some words that form sentences which would save the trouble for all three of us:
[image]

Foppotee, 2020-01-02:
@blakadder Appreciate all you do. I did read the sentences all the way to the second period. Sorry for wasting your time, but now I feel like I need to defend myself.  If I had seen what you just posted I would have not submitted the issue. I was actually 

going to suggest the message below be updated with the discord url. What I saw was this:
[image]
I clicked Open a blank issue and got this:
[image]

ascillato, 2020-01-02:
@Blakadder 

In order to not allow blank issues, you need to add the following directories and file to this repository: **github/ISSUE_TEMPLATE/config.yml** like in Tasmota:

https://github.com/arendst/Tasmota/blob/development/.github/ISSUE_TEMPLATE/config.yml