blakadder/templates #297: [Site issue]

allcoolusernamesaregone, 2020-01-08:
Hi,
the template is working but not complete.
on github:
template: '{"NAME":"hama_rgb","GPIO":[0,0,0,0,37,40,0,0,38,0,39,0,0],"FLAG":0,"BASE":18}' 
-> you have 3 sliders (color rgb select, intensity of the color, brightness of the color)

my change:
template: '{"NAME":"hama_rgb","GPIO":[0,0,0,0,37,40,0,0,38,41,39,0,0],"FLAG":0,"BASE":18}' 

-> if gpio13 is set to pwm5 41 the bulb can not just be a rgb dimmable bulb but the warm white does work. --> you get a 4th slider for the warm white 

Thanks for your great work!

blakadder, 2020-01-08:
Yeah that's not how [lights work](https://tasmota.github.io/docs/#/Lights).  

Thanks anyway.

closed 2020-01-08 by blakadder

allcoolusernamesaregone, 2020-01-08:
Sorry, but as you state
"Alexa: .... The CT control is only present to force pure white instead of RGB white. Changin CT will have no effect."

HOW exactly can you force the pure white without a slider on the web ui with the current template?

blakadder, 2020-01-08:
you cannot! use `white` command or set up white blend mode and control white uniformly