blakadder/templates #336: Alternate template for Aisirer JH-G01B Plugs.

RigsbysCatVienna, 2020-01-31:
I raised an issue a few days ago for Aisrer UK-1 plugs sold _recently_ from the same Amazon Page as the [JH-G01B models](https://templates.blakadder.com/aisirer-JH-G01B.html):

https://www.amazon.co.uk/AISIRER-Control-Compatible-British-Standards/dp/B07Y53MBGN

I ordered from this page and received a model "UK-1" and raised an issue for this for a [new template, which has been created](https://templates.blakadder.com/aisirer_UK-1.html), thank you.

But I've gone back to the two Aisirer JH-G01B plugs that I **bought back in 2018** and although they look identical to the current Amazon JH-G01B listing I found that the template listed did not work.

**Doesn't work** with my JH-G01Bs :
{"NAME":"AISIRER","GPIO":[0,17,0,0,0,0,0,0,0,52,21,0,0],"FLAG":1,"BASE":18}
**Does work** with my JH-G01Bs :
{"NAME":"AISIRER","GPIO":[0,0,0,0,56,0,0,0,21,17,0,0,0],"FLAG":1,"BASE":18}

blakadder, 2020-01-31:
its doubtful those plugs from 2018 are in circulation any more

RigsbysCatVienna, 2020-02-05:
Happy to close ... just wanted to make a note of the info

closed 2020-02-05 by RigsbysCatVienna