blakadder/templates #349: BSD01 but your numbers do not work on it

dfgas, 2020-02-07:
I did tasmota.bin and then set the settings you have and only thing that works in the button. i then found someone else that had some other changes and the led now works but still doesn't turn power on and off. the button responds and the led does something. not sure what. lol

dfgas, 2020-02-07:
gpio4 is led1i
gpio5 is led2i
gpio12 relay1
gpio13 button1

but like i said, power actually doesn't come on.

blakadder, 2020-02-07:
did you buy the device from the provided link?

dfgas, 2020-02-07:
unsure, i just know he bought them from amazon. all of the writing on them match exactly though.

blakadder, 2020-02-07:
but hardware revisions change so you need to find out the new gpio assignments

dfgas, 2020-02-07:
how do i do that?

ruffle-b, 2020-02-12:
Just got one of these and the LED and Button are on different GPIOs to the original template.

If it's any help, here's mine:

{"NAME":"YAGALA SWB3","GPIO":[52,0,0,0,0,23,0,0,21,17,22,24,0],"FLAG":1,"BASE":18}

With this template, the USB is Relay 4 and the button toggles socket 1.

dfgas, 2020-02-12:
[image]
[image]
[image]
[image]
[image]

dfgas, 2020-02-12:
i can try that when i get home unless you think mine is different

dfgas, 2020-02-12:
on mine the button is changing the on/off in the tasmota firmware but the relay is not turning on and off. same if i use the button in the firmware too

ruffle-b, 2020-02-13:
I'm surprised there are so many variants of this power strip.

There's a guide here:

https://github.com/arendst/Tasmota/wiki/Configuration-Procedure-for-New-Devices

to working out the GPIOs for a new device. Pretty simple to follow and it's what I used when the original template didn't work.

closed 2020-04-05 by blakadder