blakadder/templates #399: Meta: Sync or merge templates repo and tasmota documentation

ThomDietrich, 2020-03-20:
Hello @blakadder, @arendst and other,

(With some curfew time to spare I am willing to tune documentation on a few devices I have lying around for quite some time.)

I helped out with the Sonoff-Tasmota documentation back when it was maintained in the GitHub wiki and love what it came to in both areas! I wanted to ask if it was ever discussed how we are going to move forward with the Template page and the Tasmota documentation. It feels like there is quite some redundancy and discrepancy between the two systems, which is quite understandable looking at how they evolved.

Objectively, the Tasmota ecosystem would benefit from a merge of both webpages. Is this in planning or do you have further thoughts on that? Please feel free to point me to an existing discussion of the matter.

Additionally: While the two repositories are still separate, in which should contributors best add devices or device related docs improvements?

All the best!

arendst, 2020-03-20:
Thomas,

the Tasmota wiki is kind of dead as we decided to move to the Tasmota Docs site. Blakadder moved most of the text over to the new site but apparently there is still a lot to do.

New info should be put there.

For further info Blakadder could chime in better.

blakadder, 2020-03-20:
> Hello @blakadder, @arendst and other,
> 
> (With some curfew time to spare I am willing to tune documentation on a few devices I have lying around for quite some time.)
> 
> I helped out with the Sonoff-Tasmota documentation back when it was maintained in the GitHub wiki and love what it came to in both areas! I wanted to ask if it was ever discussed how we are going to move forward with the Template page and the Tasmota documentation. It feels like there is quite some redundancy and discrepancy between the two systems, which is quite understandable looking at how they evolved.
> 
> 1. Objectively, the Tasmota ecosystem would benefit from a **merge of both webpages**. Is this in planning or do you have further thoughts on that? Please feel free to point me to an existing discussion of the matter.

No, that idea is not feasible since they're built on two very different platforms. There is a need to migrated the current docs to a more competent platform for documentation since the current one has a few big flaws (not being indexed by google, jumbled site search, navigation requires some creative thinking)

> 2. Additionally: While the two repositories are still separate, in **which should contributors work in** to add devices or device related docs improvements?

Tasmota Docs is supposed to be a general documentation site for Tasmota software, all the device articles are legacy from the old wiki and quite a few of them are not current anymore due to many changes. 

Tasmota Templates Repository is exactly what it says on the label. I keep a very close eye on the repository and Docs are collectively maintained by the entire team.

> 3. Last point: Is it about time to delete the outdated wiki?

Outdated wiki was kept to keep the information for older versions. I guess you could create a new discussion on its importance, personally I don't find it useful and quite detrimental since its still the top google result search but since the Docs aren't indexed deleting it would not be good move

ThomDietrich, 2020-03-20:
Hey guys,
Great! Sounds like we agree on the general topics.

@blakadder you say that merging is not feasible but then go into the technical details. However, do we agree that bringing them together would be the perfect end result? I couldn't quite pull this answer from your response...

Content from both the documentation and the template page is written in markdown. The only difference is the static site generator. Jekyll is an excellent tool for that and I can see that you are already using it for the template repository. What are the concerns stopping you to include the documentation articles in the same project? I would be happy to help as I have some years of experience maintaining the openHAB documentation before we decided to move to vuepress. The documentation for home assistant is also a great example for good Jekyll mastery.

Irrespective of this, focusing purely on content: What I got from the discussion is that device specific documentation should reside in the templates repository primarily (which would stay true when the two are merged). Therefore I suggest to make this clear to the community and start working in this direction:

1. Point creators of device-specific pull requests from the documentation to the templates repository (do not accept these PRs in the documentation)
2. Migrate and delete (empty out and link) device-specific articles from documentation to the templates repository (Should be rather quick for 90% of the articles)
3. Empty out and link all articles in the old wiki (to be completely replaced as soon as google indexing works)

Wdyt?

blakadder, 2020-03-20:
> Hey guys,
> Great! Sounds like we agree on the general topics.
> 
> @blakadder you say that merging is not feasible but then go into the technical details. However, do we agree that bringing them together would be the perfect end result? I couldn't quite pull this answer from your response...

since the projects were separated for so long, templates site is a known entity and functions very well as is and considering the hours involved into transitioning and integrating with a potential new docs site i do not plan on merging the two together

> Content from both the documentation and the template page is written in markdown. The only difference is the static site generator. 

yeah... not true

> Jekyll is an excellent tool for that and I can see that you are already using it for the template repository. What are the concerns stopping you to include the documentation articles in the same project? 

first and foremost man hours needed to do so which can be contributed towards building an even better documentation site and i prefer having total control over the templates site due to some bad experiences with communal wiki editing

> I would be happy to help as I have some years of experience maintaining the openHAB documentation before we decided to move to vuepress. The documentation for home assistant is also a great example for good Jekyll mastery.

If you're eager to help the priority is finding a good platform, then architecting and designing a better documentation site which still has dynamically generated navigation while keeping ease of editing for a common user that wishes to contribute. 

> Irrespective of this, focusing purely on content: What I got from the discussion is that device specific documentation should reside in the templates repository primarily (which would stay true when [...]

Good luck with that! 

> 2. Migrate and delete (empty out and link) device-specific articles from documentation to the templates repository (Should be rather quick for 90% of the articles)

Deletion was ruled out when docs migrated from github wiki to docsify. Theo requested that all the existing information stays preserved under the Tasmota domain.

> 3. Empty out and link all articles in the old wiki (to be completely replaced as soon as google indexing works)

As soon as google indexing works? how do you plan to accomplish that?

> Wdyt?

see above ;)

ThomDietrich, 2020-03-20:
> Theo requested that all the existing information stays preserved under the Tasmota domain.

I can understand that. In the long run both areas (firmware documentation and device documentation incl. templates) should be hosted under one domain - the tasmota domain. I understand the argument by @arendst and assumed that this is the goal here.

What I laid out is **one possible plan** to resolve the redundancy and discrepancy between the THREE systems that exist atm. Shouldn't that be a common goal?
Would you like to propose a different path?

All aside, we are all voluntary open source enthusiasts willing to provide something amazing to a tinkerer community free of charge. I would love if we could discuss in a less condescending tone. Thank you

blakadder, 2020-03-20:
> > Theo requested that all the existing information stays preserved under the Tasmota domain.
> 
> I can understand that. In the long run both areas (firmware documentation and device documentation incl. templates) should be hosted under one domain - the tasmota domain. I understand the argument by @arendst and assumed that this is the goal here.
> 
> What I laid out is **one possible plan** to resolve the redundancy and discrepancy between the THREE systems that exist atm. Shouldn't that be a common goal?
> Would you like to propose a different idea?
> 
> All aside, we are all voluntary open source enthusiasts willing to provide something amazing to a tinkerer community free of charge. I did not intend to lessen all the work you've done so far and would love if we could discuss in a less condescending tone. Thank you

as i said, the focus and initial effort should be directed towards making the software documentation side best possible. templates are functioning as they are and i don't see the need to devote additional resources besides my daily involvement in maintaining the site. the proposition of merging the two was declined when it was still early and fairly simple to do so.

ThomDietrich, 2020-03-22:
Okay. Let's leave this aside for the moment.

----

Do you want to talk strategy for how to sync three device-related articles into one? An easy example is the Sonoff 4CH Pro. Documentation is available in all three places:

* https://github.com/arendst/Tasmota/wiki/sonoff-4ch-pro
* https://tasmota.github.io/docs/#/devices/Sonoff-4CH-Pro
* https://templates.blakadder.com/sonoff_4CH_Pro.html

As additions/improvement/corrections are only applied to one of these, it is inevitable to get async here. The majority of devices is maintained in the Tasmota Templates Repository (a more fitting name might be "Tasmota Devices Catalog" or similar) and therefore it should **imho** host the main articles. The articles in wiki and documentation should then only point to the main article. These articles will guide users coming from google or following their habits, with the option to be safely removed at some point.

As I said before, I would be happy to help with the legwork but we need to agree on a strategy first.

blakadder, 2020-03-22:
> Okay. Let's leave this aside for the moment.
> 
> Do you want to talk strategy for how to sync three device-related articles into one? An easy example is the Sonoff 4CH Pro. Documentation is available in all three places:
At the peril of repeating myself:
 
> * https://github.com/arendst/Tasmota/wiki/sonoff-4ch-pro

old documentation for pre v7 Tasmota. not editable

> * https://tasmota.github.io/docs/#/devices/Sonoff-4CH-Pro

New documentation site, transferred from the above link and currently maintained. Requested to remain a part of the site by Theo

> * https://templates.blakadder.com/sonoff_4CH_Pro.html

Also information copied from the first wiki with the intention of Templates being the main device how to site but per request of the main Tasmota develop scrapped in favor of https://tasmota.github.io/docs/#/devices/Sonoff-4CH-Pro

> As additions/improvement/corrections are only applied to one of these, it is inevitable to get async here. The majority of devices is maintained in the Tasmota Templates Repository (a more fitting [...]

You're trying to conflate template.blakadder.com and tasmota.github.io/docs when they're obviously and apparently two very different sites with different strategies and editing policies.

ThomDietrich, 2020-03-22:
I've been very patient and friendly with you. Sadly you ignored all my inquiries and suggestions and stay stubborn instead of working together to improve the status quo. This is not how you interact with an open source community.

@arendst I was willing to offer my experience and time to sort out the shortcomings of the current documentation page and to bring it in harmony with the templates library. I made it very clear that there are various options, I think it is about time you step in to make a call.

closed 2020-03-22 by ThomDietrich

arendst, 2020-03-23:
Regarding the 4CH-Pro example I agree with Thomas. We have to decide to move all device specific into one place while keeping legacy information also available in that same place.

What I tried to convey was that legacy 4CH info should be made available in the same location as the latest information. So the most obvious place currently would be the Template site.

Regarding the current three locations, the old wiki is just that. It has been marked as obsolete and users are linked to the new docs site. Here some intro text should link the user to the template site where ALL information regarding devices the 4CH should be revealed.

The docs site can still provide information about sensors and features as it does now.

With that new paradigm (What goes where) we must be able to use the resources of Thomas and others to make a better user experience. 

In the future we will need better search posibilities in the both the docs and template sites which may mean to redevelop the info on a seperate webenvironment someone comes up with.

Hope that keeps us all in the (documentation) boat.

arendst, 2020-03-23:
I just visited the sites to solve a Arilux issue and I must say the way it worked was fine with me.

- I searched the device in the templates site
- For more information I was correctly linked to the docs site where I found the info I needed.

So I must say. Leave it as it is for now and forget my reply earlier where I wanted to integrate device info into template.