blakadder/templates #402: Feit Dimmer Switch Template

technowizard79, 2020-03-25:
The Feit Electric Dimmer Switch has known stability issues with Tasmota, which render it more or less unusable with that firmware.  The newly posted template unfortunately doesn't correct it.  There's a whole thread on HA Community forum discussing it:

[https://community.home-assistant.io/t/costco-feit-smart-dimmer-tuya-convert-tasmota/151390](url)

Until someone can figure out the On/Off issue it has with Tasmota, this template should probably be pulled.

blakadder, 2020-03-25:
I'm not aware of any stability issues nor did the template submitter mention any. What is the link to the HA forum thread?

technowizard79, 2020-03-25:
It's in my OP, but I put it in the wrong spot under the "Feit Electric" instead of" HA Community" like I intended... sorry, it was super late when I posted it.  Here it is again:

[https://community.home-assistant.io/t/costco-feit-smart-dimmer-tuya-convert-tasmota/151390](https://community.home-assistant.io/t/costco-feit-smart-dimmer-tuya-convert-tasmota/151390)

qwertangel, 2020-04-05:
I have this dimmer,  had lot of issues with Tasmota so was forced to switch to esphome (even though I don't have HA). 
I posted the issue here ( https://github.com/arendst/Tasmota/issues/7926 ).   For some reason, mcu in that node stops responding to tasmota (for whatever reason). Unfortunately, log's don't tell much.

technowizard79, 2020-04-06:
It's far more stable on ESPhome, but it's got issues there too... push things too rapidly (e.g. the brightness buttons), and it'll intermittently drop connectivity for a few minutes (no heartbeat on log).  If memory serves me, in my experimenting with Tasmota I found that it didn't do the Off/On/On/Off bug until AFTER you set the TuyaMCU 21 option.   BTW this bug is present in as far back Sonoff/Tasmota v6.x.x.  Wanted test it with core 2.3.0, and also to see if it really was necessary to serial flash instead of OTA flash to downgrade to pre v8.x.x (it's not btw, but I can see now why they say that and block user from trying by default).  No matter which core/Tasmota version I threw at it, it still did the bug once the TuyaMCU 21 dimmer option and/or the dimmerrange is set.

Really prefer Tasmota over ESPhome, so I hope someone figures this out eventually.  Beside being amazingly priced esp82xx wifi switches (only $10-$12 each when Costco does their instant rebate sale), but they're also 3-way switches too!  Been switching (no pun intended) to Martin Jerry dimmers, and I'm having to double up to replace my 3-way switches (MJ 3-ways don't use esp chipsets sadly), where before I could just use a single Feit and an old school 3-way toggle.

blakadder, 2020-04-06:
This would require debugging by tracking serial messages from MCU to ESP by connecting to TX/RX pins... if no messages are sent when the button is pressed then its a switch+MCU problem, if the messages are sent but Tasmota doesn't read them all, or discards some of them there's a possible fix.
Best would be if you can do the same for comparison on a device that's running the original firmwaer

qwertangel, 2020-04-07:
@technowizard79 , with esphome, issue seems to be that if two messages are sent back to back then second message kills the first message (or it could be mcu related issue). given that heartbeat is sent automatically every x seconds, there is a chance that it will kill your message.
 I was able to get around this issue by sending two messages back to back with 50 ms gap. that ensures that atleast one message makes it in. (it won't work for toggle messages, just for ON/OFF/dim value).

btw this switch is amazing - it is cheap (cheaper than most of regular 1 way switch), it has dimmer, 3 way and on top of it, in 3 way configuration, it works with regular switch on other end. Given that it is bought from costco, with their return policy, you basically have unlimited warranty.

closed 2020-09-27 by blakadder