blakadder/templates #403: New WL-SW01_10 revision

rradar, 2020-03-25:
Hello,

I recently (February 2020) bought a couple of [WL-SW01_10](https://templates.blakadder.com/WL-SW01_10.html) from two different sellers and found out that they changed some things internally.

I guess the worst part is that they swapped the vertical breakout board and therefor the RX/TX pins are now hardly accesible because they are hidden behind a big elco.

[image]

[image]

[image]

It looks like there are more pin outs on the back side but how to trace them and if RX/TX is available - I don't know?

blakadder, 2020-03-25:
Trace with a multimeter in continuity mode to the esp pins

rradar, 2020-03-25:
Thank's @blakadder I could trace the VCC/GND Pins and the one labeled 'R' and 'K' on the vertical breakout board. Don't know what the later two are for?

Still no luck with the most important: RX/TX

blakadder, 2020-03-25:
TX and RX are most likely not on the main PCB, R is relay, K i don't remember what it was for, most likely Key aka button

you can do microsoldering on the chip for tx and rx or desolder the esp module board to get access to the pads

rradar, 2020-03-25:
Thank's @blakadder for the help. I guess easiest is to de-solder the big elco to gain access to the RX/TX pads on the back of the breakout board or maybe even handcraft a little tool to connect the pins even with the elco in place. I will try and report back. The documentation can then later be updated accordingly my findings...

rradar, 2020-03-26:
So my solution looks like this: Crab on of this long, stackable pin headers (the wemos d1 mini's include this ones) and bend two of them a little to fit with the RX/TX pads (I took the four pins in the middle out):

[image]

Next step is just to push them a bit against the (isolated) elco and you have nice contact to flash your thing with the usual dupont cables:
[image]

closed 2020-04-04 by rradar