blakadder/templates #413: Wrong template information for device mirabella_genio_I002608

thedeh, 2020-04-01:
Using the configuration listed under this template (with SetOption37 31 enabled) the device is unresponsive to any control commands (toggle, brightness, hue, etc.). I have confirmed the model number. All I get is a blue light when using this template. Are there any other configuration settings I've missed here?

blakadder, 2020-04-01:
everything submitted is listed on the page

obodnikov, 2020-04-12:
Right template is
{"NAME":"BQ05 Valve","GPIO":[17,255,255,255,255,255,0,0,21,157,255,255,255],"FLAG":1,"BASE":18}

blakadder, 2020-04-13:
> Right template is
> {"NAME":"BQ05 Valve","GPIO":[17,255,255,255,255,255,0,0,21,157,255,255,255],"FLAG":1,"BASE":18}

i don't think this is even remotely related to the same device

closed 2020-04-25 by blakadder