blakadder/templates #420: GD-30W 300ml Aromatherapy Diffuser config information wrong.

SirGoodenough, 2020-04-06:
Rule 1 has an instance of ```off``` that should be ```OFF``` for the device to work in home assistant.  As is HA cannot turn the light off....

Current Ru1e 1 is:
```
Rule1 on TuyaReceived#data=55AA000700056E040001007E do publish2 stat/GD-30W/EFFECT rgb_cycle endon on TuyaReceived#data=55AA000700056E040001017F do publish2 stat/GD-30W/EFFECT color endon on event#rgb_cycle do tuyasend4 110,0 endon on event#color do tuyasend4 110,1 endon on event#ON do backlog tuyasend4 110,1; tuyasend 11,1 endon on event#off do tuyasend1 11,0 endon on power3#state=1 do tuyasend4 103,1 endon on power3#state=0 do tuyasend4 103,0 endon
```
In the middle of that, the part that reads;
```on event#off do tuyasend1 11,0 endon```

shoud read:
```on event#OFF do tuyasend1 11,0 endon```

For support with Tasmota flashing and configuring the device visit [Tasmota Discord](https://discord.gg/Ks2Kzd4).
For support with Tuya-Convert please use their [GitHub Issues](https://github.com/ct-Open-Source/tuya-convert/issues).

blakadder, 2020-04-06:
fixed, thanks

closed 2020-04-06 by blakadder

bachoo786, 2020-05-15:
> fixed, thanks

Hi where can I find the fixed version and has it been tested? thanks.

blakadder, 2020-05-15:
here https://templates.blakadder.com/GD-30W... test by the OP

bachoo786, 2020-05-18:
@blakadder thanks for that :)
quick question using the template you have provided does it automatically switch off the diffuser when the water level is low? or do I have to make an automation in my homeassistant to turn off the diffuser when the water level is low? 
thanks

bachoo786, 2020-05-18:
I just tried the update template and I can report that I cannot control the lamp with Homeassistant also when I click the buttons on the diffuser the state doesnt change in homeassistant.

blakadder, 2020-05-19:
then you didn't follow the instructions

bachoo786, 2020-05-19:
> then you didn't follow the instructions

Could you elaborate more please?

blakadder, 2020-05-19:
When you open the template there's text below the template window. You're supposed to read it and issue the given commands in Tasmota

bachoo786, 2020-05-19:
I have already set the template to the correct one (see image) 
[image]

I am using maxcio Air diffuser and as per your template you have mentioned it is the same as GD-30W. So I copied all the commands on the GD-30W page and put them in my console i.e. all the rules etc. 

However many people who are using home assistant have complained that the lamp for example doesn't store the state in home assistant if you turn it on on Home assistsnt it quickly goes back to the off state but on the diffuser it remains on. 

Also my other question was that does your template turn the diffuser off when there is no water? Or do I have to make an automation on home assistsnt to do that? 

Thank you

blakadder, 2020-05-19:
> I have already set the template to the correct one (see image)
> I am using maxcio Air diffuser and as per your template you have mentioned it is the same as GD-30W. So I copied all the commands on the GD-30W page and put them in my console i.e. all the rules etc.

Sorry, no experience with Maxcio diffuser, that one was submitted by someone else.
 
> However many people who are using home assistant have complained that the lamp for example doesn't store the state in home assistant if you turn it on on Home assistsnt it quickly goes back to the off state but on the diffuser it remains on.

Many people complain about many things but the setup worked in my system.

> Also my other question was that does your template turn the diffuser off when there is no water? Or do I have to make an automation on home assistsnt to do that?

No, the MCU turns it off

> Thank you