blakadder/templates #431: BQ05 wrong template information

obodnikov, 2020-04-12:
Wrong template information on https://templates.blakadder.com/irrigation_timer_BQ05.html
Completely not work
Right one is:

{"NAME":"BQ05","GPIO":[17,255,255,255,255,255,0,0,21,157,255,255,255],"FLAG":1,"BASE":18}

Please update page

For support with Tasmota flashing and configuring the device visit [Tasmota Discord](https://discord.gg/Ks2Kzd4).
For support with Tuya-Convert please use their [GitHub Issues](https://github.com/ct-Open-Source/tuya-convert/issues).

blakadder, 2020-04-13:
it worked for others, please submit a new v2 template

closed 2020-04-13 by blakadder

s00500, 2020-07-19:
Hey everyone, I agree, the template info was not working for me, using the template provided by this issue it is working fine!