blakadder/templates #489: KMC 70113 Template Issue

adrianrego, 2020-05-23:
I purchased a 3 pack of these bulbs from Amazon (link in template page) and successfully flashed all 3 with Tasmota via Tuya-convert.  Model on the box says 70113WC.  However, I believe the template has a few issues.

Template:
https://templates.blakadder.com/kmc_70113.html

Issues:

- Tasmota status is inverted.  When Tasmota displays an OFF status, the bulb is ON and vice-versa.  I've tried flipping PWN1/2 to their inverted options (PWN1i/PWN2i) and have had no luck.

- Tasmota does not provide controls for dimming or changing the color temperature of these bulbs.

2 of the bulbs are running Tasmota 8.3.1 and the other is running 8.1.0.

Any thoughts?  I'm new to this and can't find any information information on these bulbs online.

blakadder, 2020-05-23:
sound like you didnt apply the template at all

adrianrego, 2020-05-23:
It's applied as far as I know.  I can toggle the light on/off.

[image]
[image]

blakadder, 2020-05-23:
it is not applied

adrianrego, 2020-05-23:
How do I apply it then?  I went into Configure Other, pasted in the template, checked Activate, and hit Save.

adrianrego, 2020-05-23:
Ok, I see what you mean here.  It's still showing Sonoff Basic Module.  Could there be an issue where the firmware isn't applying/saving the template?

adrianrego, 2020-05-23:
Seems like it's an issue with the device.  Can't change the module.  Bulb looks like it goes into a bootloop and eventually ends up resetting to default module. Will open an issue in the Tasmota project if I can't make any progress.

Thanks for your help.

closed 2020-05-23 by adrianrego