blakadder/templates #525: Arlec MAL300HA - No Manual Override

danielhelmstedt, 2020-06-16:
Hey,
After disassembling my Arlec Smart Security Sensor Light (MAL300HA) some while back and being unable to reverse-engineer the Manual Override function, I have found your template is the same, only able to get Sensor Mode on/off.

The TYWE2S module has:
VCC (Blue)
GND (White)
GPIO4 (Red) -controls the sensor mode
GPIO5 (Brown) - LED
GPIO12 (Black) - connected to.... It seems to also run to the IC that handles the PIR/Pots/Relay, but toggling it doesn't function as expected. I even tried jumping R4.

You can see below I've traced the PCB path. The PIR goes into the IC through the pin labelled B, GPIO12 ends at Pin A
[image]
[image]
[image]

Hopefully someone else may have some insights on this?

bigrizzo, 2020-06-27:
I bought a couple of these yesterday. I flashed one with Tasmota and used the template giving only the sensor on/off function in the hope I could find a way to manually control the lights but no luck. I pulled off the heatshrink and found the TYWE2S module and that's as far as I got. I didn't see the modules in your posts. Are they inside the motion sensor? In any case, I'm with you. I'd love to be able to control the light via Tasmota. Hoping someone will pitch in on this. I'll be keenly watching. Thanks for posting.

danielhelmstedt, 2020-07-04:
Yes, the PCBs are inside the sensor housing. I'm not sure where to go on this.

bigrizzo, 2020-09-03:
> Yes, the PCBs are inside the sensor housing. I'm not sure where to go on this.

Hi Daniel. Did you ever make any progress on this? I'm still waiting for someone smarter than me to figure it out.

danielhelmstedt, 2021-01-02:
I've finally had a bit of a chance to get back to this. I have some findings, but I haven't really got anywhere with them. The TYWE2S module is connected to (I believe) a 556 timer chip (Labelled with 21A but I can't find anything on that. It seems to fit the pinout of a 14pin 556), but only using one half of it, so essentially a 555.

"A" and "B" are labelled both on the TYWE2S board and on the sensor board next to the 556, so that's nice and convenient. GPIO5 however seems to not only run on top of the sensor board to the LED, but it also runs on the bottom of the board over to the Control Voltage pin of the 555.

GPIO4/Red - “B” - Control Voltage A
GPIO5/Brown - LED, Threshold A
GPIO12/Black - “A” - Trigger A
VCC - Blue
GND - White

[Some reading on 555 timers](https://www.electronics-tutorials.ws/waveforms/555_timer.html) tells us what these do

> **Control Voltage,** This pin controls the timing of the 555 by overriding the 2/3Vcc level of the voltage divider network. By applying a voltage to this pin the width of the output signal can be varied independently of the RC timing network. When not used it is connected to ground via a 10nF capacitor to eliminate any noise.

> **Threshold**, The positive input to comparator No 2. This pin is used to reset the Flip-flop when the voltage applied to it exceeds 2/3Vcc causing the output to switch from “HIGH” to “LOW” state. This pin connects directly to the RC timing circuit.

>  **Trigger**, The negative input to comparator No 1. A negative pulse on this pin “sets” the internal Flip-flop when the voltage drops below 1/3Vcc causing the output to switch from a “LOW” to a “HIGH” state.

bigrizzo, 2021-01-05:
Nice work, Daniel. Sounds like progress. I hope you're able to make sense of it all to extract the full functionality in Tasmota. Mine are still just working in 'dumb' sensor mode until someone cracks the mystery. Still way beyond my skillset.

Lenbok, 2021-04-24:
In case it is of interest, I've got the newer model MAL315HA working pretty nicely. The hardware looks a bit different, as it's using TuyaMCU, but at least that lets Tasmota adjust PIR sensitivity, lux, duration etc. See #1023

bigrizzo, 2021-04-24:
> In case it is of interest, I've got the newer model MAL315HA working pretty nicely. The hardware looks a bit different, as it's using TuyaMCU, but at least that lets Tasmota adjust PIR sensitivity, lux, duration etc. See #1023

Hi Lenbok. I'd be interested to know how you got it working. Any reason why it wouldn't work with the older MAL300HA?

bigrizzo, 2021-04-24:
> In case it is of interest, I've got the newer model MAL315HA working pretty nicely. The hardware looks a bit different, as it's using TuyaMCU, but at least that lets Tasmota adjust PIR sensitivity, lux, duration etc. See #1023

Nice work, mate! I just jumped over to your writeup and it's well beyond my level of expertise. I have the MAL300HA and I've flashed it with Tasmota. I know how to get into console and enter commands. I'm just not sure if what you've written up for the MAL315HA works with the MAL300HA? You mention the TuyaMCU and I see commands in your writeup to enable it. So you're saying the MAL300HA doesn't have the TuyaMCU? Sorry mate. If you have any advice about how I can get the MAL300HA going, I'd be very grateful.

Lenbok, 2021-04-24:
Yeah, looking at the PCB picture in this issue vs mine it looks like the hardware is significantly different, using the communications to the external TuyaMCU and using software adjustment vs the physical potentiometers in the MAL300HA. My comment was more of a heads-up (as I originally thought the model I was purchasing from bunnings was the MAL300HA).