blakadder/templates #569: Question: Determine WiFi Module in teckin_SP22

xxcite, 2020-07-06:
Hi,

regarding the [teckin_SP22](https://github.com/blakadder/templates/blob/master/_templates/teckin_SP22) is there a way to determine, what kind of WiFi module is in the smart plug?

Thanks.

blakadder, 2020-07-06:
yes, open it up and visually inspect it

xxcite, 2020-07-06:
Thought there is another way to check without opening it :-D

Thanks anyway

blakadder, 2020-07-06:
nothing 100% accurate. you can try the following

if it runs tuya you can try to tuya convert it and check logs for the MAC address or the dreaded "not ESP8266 chip" message

or 

let it connect to your wifi network with original firmware, grab its MAC address and use https://dnschecker.org/mac-lookup.php to determine the manufacturer of the module

xxcite, 2020-07-06:
Thanks for your help. The DNS checker showed Vendor/Company "Hangzhou Aixiangji Technology Co., Ltd" which does not help I guess. I will try the other option.

closed 2020-07-06 by xxcite