blakadder/templates #586: brilliantsmart_20967 DimmerRange command

cojaxx, 2020-07-17:
Hello,
Came across your post with the template for a brilliant smart dimmer.

Got it to work except for this line of code.

15:19:57 CMD: DimmerRange 1,255
15:19:57 MQT: dimmer_switch/stat/RESULT = {"Command":"Unknown"}

Just wondering if there is something I have missed?

Regards,
Peter

mgiggs, 2020-07-21:
You need a space after the ,

I would recommend DimmerRange 30, 100

Going to 255 makes no difference and below 30 turns it off.

cojaxx, 2020-07-21:
Hello,
Thanks! It still didn't work but I re-entered these two commands then it started working.

TuyaMCU 11,1
TuyaMCU 21,2

They must not have applied properly the first time.

All working now :) Thanks

cojaxx, 2020-07-21:
Oh you might want to update the documentation for that template as there is no space after the , 

https://templates.blakadder.com/brilliantsmart_20967.html