blakadder/templates #590: [Site issue] WF-DS01 template not offering dimming

giceton, 2020-07-21:
Hi. I have the WF-DS01 and having used the template the device does not offer a dimming function, just toggle on / off. Please let me know if I can help with this. Thanks.

blakadder, 2020-07-22:
Are you offering help or seeking help?

Did you follow **all of the instructions** ??

giceton, 2020-07-23:
Hi. Apologies for not being clear! I was asking for help please as I have this device but its not working as expected (i.e. there is no dimming option).

blakadder, 2020-07-23:
while apologizing you missed answering to the question i asked

giceton, 2020-07-23:
Yes, I followed the instructions. However, I have just now tried to enter the commands separately as in:

Backlog Ledtable 0
TuyaMCU 21,2

rather than all in one line as in the instructions and this has worked. I'm just going to check the maximum light level is correct as it now seems dimmer than it was. I have another device I will now check this on. Thanks for your efforts - I sent you a small offering via paypal..

giceton, 2020-07-23:
FYI I had to change the dimmer range to 11, 255. All now works well. Thanks.

closed 2020-07-23 by giceton