blakadder/templates #642: Missing full IR rules for LSC LED strip

adam-kulagowski, 2020-08-26:
Hi,

The rules for https://templates.blakadder.com/lsc_smart_connect_led_strip.html are incomplete. Below are complete rules for IR remote:

```
Rule1 ON IrReceived#Data=0x00F740BF do power off ENDON ON IrReceived#Data=0x00F7C03F do power on ENDON ON IrReceived#Data=0x00F700FF do dimmer + ENDON ON IrReceived#Data=0x00F7807F do dimmer - ENDON ON IrReceived#Data=0x00F720DF do color2 ff0000 ENDON ON IrReceived#Data=0x00F7A05F do color2 00ff00 ENDON ON IrReceived#Data=0x00F7609F do color2 0000ff ENDON

Rule2 ON IrReceived#Data=0x00F710EF do color2 ff5000 ENDON ON IrReceived#Data=0x00F7906F do color2 aeff00 ENDON  ON IrReceived#Data=0x00F750AF do color2 0077ff ENDON ON IrReceived#Data=0x00F730CF do color2 f2ba41 ENDON ON IrReceived#Data=0x00F7B04F do color2 41e3f2 ENDON ON IrReceived#Data=0x00F7708F do color2 a80355 ENDON ON IrReceived#Data=0x00F708F7 do color2 fcce62 ENDON ON IrReceived#Data=0x00F78877 do color2 25a9c4 ENDON ON IrReceived#Data=0x00F748B7 do color2 c4254a ENDON

Rule3 ON IrReceived#Data=0x00F728D7 do color2 fcfc05 ENDON ON IrReceived#Data=0x00F7A857 do color2 2d7c93 ENDON ON IrReceived#Data=0x00F76897 do color2 fc5582 ENDON ON IrReceived#Data=0x00F7E01F do color2 ffffff ENDON ON IrReceived#Data=0x00F7E817 do scheme 2 ENDON ON IrReceived#Data=0x00F7C837 do scheme 3 ENDON ON IrReceived#Data=0x00F7F00F do scheme 4 ENDON ON IrReceived#Data=0x00F7D02F do Power 3 ENDON
```

blakadder, 2020-08-26:
added! thanks

adam-kulagowski, 2020-08-27:
Also: 
The commit a42541e ( @norbert-walter ) makes the template incompatible w/ RGBW LSC LED (older model supported by tasmota):

The new template is only RGB+CT+WW model (which requires re-soldering).

In result the legacy model had White channel disabled. I believe, that this distinction should be noted in documentation.

blakadder, 2020-09-27:
restored the original template

closed 2020-09-27 by blakadder