blakadder/templates #674: LSC Smart Power plug v.1 now also contains a WB2S module instead of an esp module

FransOv, 2020-09-24:
LSC Smart Power plug v.1 (Art.no. 2578685 970761 v1.0) now also contains a WB2S module instead of an esp module.
[image]

blakadder, 2020-09-24:
how is that plug different from the one listed? https://templates.blakadder.com/lsc_smart_connect_power_plug.html

FransOv, 2020-09-24:
This is a version 1 plug (with screws) and still anWB2S module instead of an esp module. As written now on https://templates.blakadder.com/lsc_smart_connect_power_plug.html, the warning seems to apply only to version 2.

blakadder, 2020-09-27:
there's an edit feature for each page so it is easy to correct erroneous information

closed 2021-09-07 by blakadder