blakadder/templates #686: Gosund UP111 Template no longer allowing the physical power button to run

rssfed23, 2020-10-02:
The template at https://templates.blakadder.com/gosund_UP111.html has worked fine for many Tasmota versions for myself.
However, since updating to 8.5.0 (issue also exists on 8.5.1), the physical power button no longer works to turn the plug on/off when pressed.
All other functions of the plug seem to work fine and can be turned on/off via the Tasmota web UI or MQTT. But when pressed the light will flash on/off but not actually trigger the switch to turn on/off. 

Current template used for reference:
```https://templates.blakadder.com/gosund_UP111.html```

blakadder, 2020-10-02:
try `setoption73 0`

rssfed23, 2020-10-02:
That did the trick! - Thanks I'll close this

closed 2020-10-02 by rssfed23