blakadder/templates #762: [Site issue] - Loratap SC411WSC-EU now uses realtek chip

sthorley, 2020-11-20:
Hi can the page for the Loratap SC411WSC-EU please be updated to show that recently manufactured versions are now using the Realtek chip and are not able to be used with Tasmota. I explicitly checked that the SC411 had a tasmota template before buying however when I went to flash it I discovered a WR3E chip inside.

Images here: https://imgur.com/a/unTRZZN

sthorley, 2020-11-20:
In fact if you look at the second aliexpress link it now shows in the specifications that it uses the realtek chip:

"Specification

Input Voltage: 100-250VAC, 50/60Hz
Max Load: 600W
Works with: 4 wires(Earth Wire, Netural Wire, L1, L2) AC motor
App Name: Smart Life, Tuya APP
Voice Control: Amazon Alexa, Google Assistant
Wireless Protocol: Wi-Fi 2.4GHz(can't work with 5Ghz Wi-Fi Band)
**Main Chip: REALTKE RTL8710BN Inside**
Dimensions: 86mm*86mm*35mm"

blakadder, 2020-11-20:
Yes, seems it's become a common occurence with LoraTap devices to use both

PR #771 (2020-11-23): Update LotaTap SC411WSC-EU page with updated hardware warning.
As per #762 - update template page with potential hardware issue.
Fixes #762
closed 2020-11-23 by blakadder