blakadder/templates #797: [Site issue] Gosund WP3 v2 Template none-functional

nate2014jatc, 2020-12-08:
Hello,
I would like to report that the v2 template is not working currently with the Gosund WP3 plug at this time.
By looking at the template in github, I was able to get the `template` instead of the `template9` and this configuration works.

Is this an error with the documentation parsing for the template site, or perhaps an accidental omission?

blakadder, 2020-12-08:
the system selects a v9 template over the old template

nate2014jatc, 2020-12-09:
Alrighty then, 
I would still like to report that the template the site is presenting isn't working.

blakadder, 2020-12-09:
Do you have an alternate one?

nate2014jatc, 2020-12-09:
One that's listed as `template` instead of `template9` is working fine.

blakadder, 2020-12-09:
and what is the template9 version of it?

nate2014jatc, 2020-12-18:
Looks like you fixed this with commit `4802a0925c1acc2dad4023c0e59a897f9ba04afc`.

closed 2020-12-18 by nate2014jatc