blakadder/templates #837: [Site issue] amzdest_c158: second plug not working when turned on?

vinhdizzo, 2020-12-30:
I'm not sure where is the proper place to report this, but I just added the template for this device. The button on the device is able to turn outlet 1 on and off. I'm also able to toggle outlet 1 on and off via the Tasmota page. However, toggling outlet 2 does not turn on the second outlet physically, even though the Tasmota page shows on.

Thank you.

blakadder, 2020-12-30:
Then the template is not correct for your model.  https://tasmota.github.io/docs/Configuration-Procedure-for-New-Devices/

vinhdizzo, 2020-12-30:
Thanks for this.

Using the suggested tutorial, I can confirm that relay should be on GPIO14 instead of GPIO3. This is what I am currently using. What is the proper method to get this new template to replace the old template on the site?

```
{"NAME":"Amzdest c158","GPIO":[0,32,0,0,2720,2656,0,0,224,2624,225,0,0,0],"FLAG":0,"BASE":18}
```

blakadder, 2020-12-30:
i will adit in the next edit session!

closed 2021-09-07 by blakadder