blakadder/templates #1013: Is the SS311KWS AliExpress link still correct ?

demondev, 2021-04-19:
This page https://templates.blakadder.com/kinetic_switch_SS311KWS.html links this AliExpress product https://www.aliexpress.com/item/32953812650.html 

The image of the product on blakadder looks nothing like that given on AliExpress. Also the switch pic on AliExpress looks a lot like this unsupported item https://templates.blakadder.com/unsupported/loratap_SP311WST.html

blakadder, 2021-04-20:
it is submitted that way

demondev, 2021-04-20:
You are saying a third party provided this information to you and you created the template page ? Which implies you have no way of verifying this information ? 

It very much looks to me like the AliExpress seller has edited the item since the template page was created and if you bought the item now it would not be incompatible. There is no mention of the SS311KWS part number in the current listing,

blakadder, 2021-04-20:
Nothing new, Amazon sellers do the same thing

demondev, 2021-04-20:
What do you do with the template page in this case ? Remove the link ? Put a note about the link ?

blakadder, 2021-04-20:
if its spotted by someone the dead/wrong link will be removed as in this case

demondev, 2021-04-20:
Perfect, thanks.

closed 2021-04-20 by demondev