blakadder/templates #1029: GD-30W: strange behavior for mist strength

Nardol, 2021-04-25:
I reset my 400 ML Maxcio diffuser and applyed last rules from GD-30W diffuser.
I also applyed the configuration for Home Assistant.
If I change mist strength by pressing the button, state is right in Home Assistant, for law and high.
If I change it using the HA fan, the strength is changed right but if I switch to High mode and press the button on the device, it is not switched off, I have to press it twice to switch the diffuser off even if it is already on high mode.
I have translated the modes to be able to have texts in French, so I also changed the MQTT queries to change these.
Could it make a difference?
Or is there an unexpected behavior with the physical button press which doe not reckognize if mist strength has been changed using Home Assistant?

My Home Assistant fan configuration (in a package):
```yaml
tasmota_diffuseur_bureau:
  fan:
    - platform: mqtt
      name: "Diffuseur Bureau"
      unique_id: fan.diffuseur_bureau
      state_topic: "stat/diffuseur_bureau/RESULT"
      state_value_template: "{% if value_json.POWER1 is defined %}{{ value_json.POWER1 }}{% endif%}"
      command_topic: "cmnd/diffuseur_bureau/POWER1"
      preset_modes:
      -  "Fort"
      -  "Faible"
      preset_mode_state_topic: "stat/diffuseur_bureau/fanmode"
      preset_mode_command_topic: "cmnd/diffuseur_bureau/tuyaenum1"
      preset_mode_command_template: '{{ 1 if value == "Fort" else 0 }}'
      availability_topic: "tele/diffuseur_bureau/LWT"
      payload_available: "Online"
      payload_not_available: "Offline"
```

My device Rules:
```console
20:26:13.850 CMD: Rule0
20:26:13.859 MQT: stat/diffuseur_bureau/RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":112,"Free":399,"Rules":"on tuyareceived#dptype4id13!=0 do tuyasend4 13,0 endon on scheme#data!=0 do backlog0 scheme 0; tuyaenum2 0 endon"}}
20:26:13.877 MQT: stat/diffuseur_bureau/RESULT = {"Rule2":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":223,"Free":288,"Rules":"on tuyareceived#dptype5id12 do publish stat/%topic%/error %value% endon on tuyareceived#dptype4id103==1 do publish stat/%topic%/fanmode Fort endon on tuyareceived#dptype4id103==0 do publish stat/%topic%/fanmode Faible endon"}}
20:26:13.895 MQT: stat/diffuseur_bureau/RESULT = {"Rule3":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
```

Maybe I do something wrong?

blakadder, 2021-04-25:
fanmode status gets changed only when its reported over a dpid, if it doesn't get reported when you press the device button you're  out of luck.

you can implement a rule to periodically send `tuyasend0` command

Nardol, 2021-04-26:
OK, so to be sure, even if the diffuser is in high mist strength it does not mean it is sure the button will act as expected?
Because it looks like it is only the button, the diffuser is in high mode as expected.

I feel strange the display does not report status as soon as the mist strength is changed, that's why I prefer to re-formulate.

blakadder, 2021-04-26:
its very simple:
dpid103==0 = mist low
dpid103==1 = mist high

if that's not the case with your device you need to change the rules to reflect that

i don't really understand what display you're referring to or what are you reformulating.

Nardol, 2021-04-26:
I changed the text as my config should show, except if there is somewhere else where I should have done it in the first post of this issue.
I tried to re-formulate what you answered me to be sure I understood everything.

I talked about what I observe on the diffuser itself.

If I change the mode from HA, when looking to what diffuser is doing, mist strength is higher.
So the command looks like OK, but if I press the physical button the diffuser does not stop.
And when pressed a second time the diffuser stops, but in Home Assistant the state is not changed.

I hop I am better understandable, sorry for the confusion.

Nardol, 2021-04-26:
After more testing Home Assistant is not the cause, I simply can summarize by telling if I switch the diffuser on using Tasmota, when I switch it off using the physical button the Tasmota switch stays On even if the diffuser is off.

blakadder, 2021-04-26:
That's a badly designed device since it doesn't seem to publish dpid updates if you use the physical button. the GD-30W does so so there is no issue with that

Nardol, 2021-04-26:
OK, so issue is from the Maxcio diffuser...
Do you see any possible worrk around or do you think the best solution is to use the previous template and home assistant configuration or simply not use the physical button?

blakadder, 2021-04-26:
If it worked with the previous configuration use that

Nardol, 2021-04-26:
After resetting my device and testing, this behavior also occurs with the vrevious configuration. But I had never noticed it before, strangely I did not test this procedure.
So OK, I'll look into checking the dpid types.

Nardol, 2021-04-26:
So I reset my device and applied the GD-30W template.
For me it looks OK but to be honest I am lost in all these values :slightly_frowning_face: I made a long read at this:
```console
18:05:16.964 {"TuyaReceived":{"Data":"55AA010700050C050001001E","Cmnd":7,"CmndData":"0C05000100","DpType5Id12":"0x00","12":{"DpId":12,"DpIdType":5,"DpIdData":"00"}}}
18:05:16.979 TYA: fnId=0 is set for dpId=12
18:05:16.984 {"TuyaReceived":{"Data":"55AA01070005670400010078","Cmnd":7,"CmndData":"6704000100","DpType4Id103":0,"103":{"DpId":103,"DpIdType":4,"DpIdData":"00"}}}
18:05:16.992 TYA: fnId=61 is set for dpId=103
18:05:16.995 SRC: Switch
18:05:16.998 CMD: Group 0, Index 1, Command "TUYAENUM", Data "0"
18:05:17.001 MQT: stat/diffuseur_bureau/RESULT = {"TuyaEnum1":0}
18:05:17.011 {"TuyaReceived":{"Data":"55AA01070005010100010110","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
18:05:17.020 TYA: fnId=11 is set for dpId=1
18:05:17.022 TYA: RX Relay-1 --> MCU State: On Current State:Off
18:05:17.024 SRC: Switch
18:05:17.030 MQT: stat/diffuseur_bureau/RESULT = {"POWER1":"ON"}
18:05:17.035 MQT: stat/diffuseur_bureau/POWER1 = ON
18:05:17.555 CFG: Saved to flash at F5, Count 23, Bytes 4096
18:05:21.181 TYA: Send "55aa00000000ff"
18:05:21.190 {"TuyaReceived":{"Data":"55AA010000010102","Cmnd":0,"CmndData":"01"}}
18:05:21.199 TYA: Heartbeat
18:05:21.588 {"TuyaReceived":{"Data":"55AA010700050C050001001E","Cmnd":7,"CmndData":"0C05000100","DpType5Id12":"0x00","12":{"DpId":12,"DpIdType":5,"DpIdData":"00"}}}
18:05:21.598 TYA: fnId=0 is set for dpId=12
18:05:21.602 {"TuyaReceived":{"Data":"55AA01070005670400010179","Cmnd":7,"CmndData":"6704000101","DpType4Id103":1,"103":{"DpId":103,"DpIdType":4,"DpIdData":"01"}}}
18:05:21.610 TYA: fnId=61 is set for dpId=103
18:05:21.613 SRC: Switch
18:05:21.616 CMD: Group 0, Index 1, Command "TUYAENUM", Data "1"
18:05:21.623 MQT: stat/diffuseur_bureau/RESULT = {"TuyaEnum1":1}
18:05:21.635 {"TuyaReceived":{"Data":"55AA01070005010100010110","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
18:05:21.646 TYA: fnId=11 is set for dpId=1
18:05:21.648 TYA: RX Relay-1 --> MCU State: On Current State:On
18:05:25.240 {"TuyaReceived":{"Data":"55AA0107000567040001027A","Cmnd":7,"CmndData":"6704000102","DpType4Id103":2,"103":{"DpId":103,"DpIdType":4,"DpIdData":"02"}}}
18:05:25.249 TYA: fnId=61 is set for dpId=103
18:05:25.252 SRC: Switch
18:05:25.255 CMD: Group 0, Index 1, Command "TUYAENUM", Data "2"
18:05:25.265 MQT: stat/diffuseur_bureau/RESULT = {"Command":"Error"}
18:05:25.277 {"TuyaReceived":{"Data":"55AA0107000501010001000F","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
18:05:25.287 TYA: fnId=11 is set for dpId=1
18:05:25.289 TYA: RX Relay-1 --> MCU State: Off Current State:On
18:05:25.292 SRC: Switch
18:05:25.298 MQT: stat/diffuseur_bureau/RESULT = {"POWER1":"OFF"}
18:05:25.303 MQT: stat/diffuseur_bureau/POWER1 = OFF
```

I typed `weblog 4"` in the console and observed what happened when I pressed the physical button.

A point which looks very strange for me is:
`18:05:25.265 MQT: stat/diffuseur_bureau/RESULT = {"Command":"Error"}`
When I switch the diffuser off, then the state looks like updating right but is not updated on the Tasmota web interface, the power button is still "on".

I only applied the template, configured MQTT&wi-fi, pasted the two backlog commands and set Rule1 since I reset my Maxcio diffuser.
In fact I stopped just before adding rules for Home Assistant.

I hop I made things right and did not miss-understood.

blakadder, 2021-04-27:
Your mist strength dpid has more than 2 values so you need to set it up differently

Nardol, 2021-04-28:
OK, I missed the third value; so I set `tuyaenum1` to have 3 values instead of 2.
Now I have no error when the device is switched off and observe the following:
- Value is 0 for low;
- 1 for high;
- 2 when the device is switched off
- But Power1 state is not updated, even with `TuyaSend0`.

As a beginner with no enough knowledge I tried to set Power1 to 0 when tuyaenum1 value was 2, which has been the worst idea I had yesterday. Fortunately power cyccle procedure to reset helped.
I think `TuyaSend1 1,0` would produce the same effect if I make a rule as I did yesterday for Power1 right?
Do you think this diffuser is definitely too badly designed to be able to have the physical button synchronized as expected?

closed 2021-09-07 by blakadder