blakadder/templates #1094: Athom Light Swiches

tungmeister, 2021-07-07:
@blakadder Hi, do you know if the red and blue LEDs on the Athom light switches  (shown here https://templates.blakadder.com/athom_SW11-TAS-3EU.html) are individually controlable? I'd like to use the switches separately from the relays for controlling smart bulbs - the switches I tried recently couldn't do this.

blakadder, 2021-07-07:
There are separate LEDs in the template so I'd say: yes

tungmeister, 2021-07-08:
thanks - I had seen the template but just wanted to make sure. Was it provided by the manufacturer?

blakadder, 2021-07-08:
yes, i have the clicky version and that one has the same template so you can control led's independently. I actually use one of the keys by mapping the Relay3 to Led3 so it only shows the state of a remote without pointlessly switching the relay

tungmeister, 2021-07-08:
thanks for that - i'll get an order in!

closed 2021-07-08 by tungmeister