blakadder/templates #1102: [Template issue] SS02 toggle state reversed for TreatLife 3-Way Switch

smizles, 2021-07-16:
The template for SS02 apparently creates two outputs/relays that appear to alternate (I believe this is to handle the 3-way functionality by switching the power from the load side to the traveler wire?). However, in the console the two relays appear to always have the same state even though they alternate in the main dashboard.

I noticed this quirk while creating an automation in home assistant that is supposed to turn off a plug when the switch is turned off. The plug's state is always inverted from the switch regardless of whether I use the main switch entity or the "switch..._<mac>_switch_relay_1" entity. Obviously it is easy to hard code the reversed state into the automation, but I'm worried about any confusion that might be caused in the user interface or other people that might be having a similar issue.
[image]

```
06:54:44.945 MQT: tele/u_living_room_lamp_switch/STATE = {"Time":"2021-07-16T06:54:44","Uptime":"0T05:38:51","UptimeSec":20331,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"SimsNoT","BSSId":"F6:92:BF:A8:4F:F5","Channel":11,"Mode":"11n","RSSI":100,"Signal":-49,"LinkCount":1,"Downtime":"0T00:01:03"}}
06:54:44.954 MQT: tele/u_living_room_lamp_switch/SENSOR = {"Time":"2021-07-16T06:54:44","Switch1":"ON"}
06:54:45.048 MQT: stat/u_living_room_lamp_switch/RESULT = {"POWER2":"OFF"}
06:54:45.053 MQT: stat/u_living_room_lamp_switch/POWER2 = OFF
06:54:45.334 MQT: stat/u_living_room_lamp_switch/RESULT = {"POWER1":"OFF"}
06:54:45.338 MQT: stat/u_living_room_lamp_switch/POWER1 = OFF
06:55:14.790 MQT: stat/u_living_room_lamp_switch/RESULT = {"POWER2":"ON"}
06:55:14.794 MQT: stat/u_living_room_lamp_switch/POWER2 = ON
06:55:15.031 MQT: stat/u_living_room_lamp_switch/RESULT = {"POWER1":"ON"}
06:55:15.036 MQT: stat/u_living_room_lamp_switch/POWER1 = ON
06:55:21.005 MQT: stat/u_living_room_lamp_switch/RESULT = {"POWER2":"OFF"}
06:55:21.010 MQT: stat/u_living_room_lamp_switch/POWER2 = OFF
06:55:21.288 MQT: stat/u_living_room_lamp_switch/RESULT = {"POWER1":"OFF"}
06:55:21.293 MQT: stat/u_living_room_lamp_switch/POWER1 = OFF
06:59:44.934 MQT: tele/u_living_room_lamp_switch/STATE = {"Time":"2021-07-16T06:59:44","Uptime":"0T05:43:51","UptimeSec":20631,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":31,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"SimsNoT","BSSId":"F6:92:BF:A8:4F:F5","Channel":11,"Mode":"11n","RSSI":100,"Signal":-50,"LinkCount":1,"Downtime":"0T00:01:03"}}
06:59:44.940 MQT: tele/u_living_room_lamp_switch/SENSOR = {"Time":"2021-07-16T06:59:44","Switch1":"OFF"}
```

For support with Tasmota flashing and configuring the device visit [Tasmota Discord](https://discord.gg/Ks2Kzd4).
For support with Tuya-Convert please use their [GitHub Issues](https://github.com/ct-Open-Source/tuya-convert/issues).

smizles, 2021-07-16:
The recommendation in this thread (https://github.com/arendst/Tasmota/issues/4268) to enter `SwitchMode 2` in the console corrects this issue, but I believe this should be set by default in the template, which it is not.

blakadder, 2021-07-16:
It will, perhaps, work better if you set `Relay` to `Relay_i` instead?

blakadder, 2021-09-07:
Did Relay_i resolve the issue?

closed 2022-04-06 by blakadder