blakadder/templates #1170: Updated info for EIE3-1001-WHT

motolav, 2021-10-05:
https://templates.blakadder.com/energizer_EIE3-1001-WHT.html

The plug uses a Common Anode RGB LED
GPIO 2 for Red
GPIO 13 for Blue
GPIO 16 for Green
All inverted outputs
The template just lists pins 13 and 16 as plain LEDs which is incorrect 

For Power Monitoring the GPIOs should be used are
GPIO 4 as HLW8012 CF
GPIO 5 as HLWBL CF1
GPIO 12 as HLWBL SELi

blakadder, 2021-10-06:
Post the new template with those changes

motolav, 2021-10-06:
The new template should be
{"NAME":"Energizer EIE3-1001-WHT","GPIO":[32,0,448,0,2688,2656,0,0,2624,450,224,0,449,0],"FLAG":0,"BASE":18}

crossan007, 2021-10-07:
@motolav that template does seem to correctly drive the LED (displaying color, saturation, and brightness sliders on the main menu page); however, the template also separates the state of the LED from the state of the main relay;  generally, the LED should indicate the state of the relay (on/off).

The HLWBL SELi adjustment does seem to properly drive the power monitoring IC to select Voltage and current (I have not re-calibrated my plug yet with a known wattage load, but the voltage monitor is working properly.)

crossan007, 2021-10-07:
It looks like this change has already been committed: https://github.com/blakadder/templates/commit/577c8e777603088fdfe2677f69ddbb35b03b93b0

(Linking so it shows up on both pages on GitHub)

motolav, 2021-10-07:
@crossan007 For the LED you just need to setup a rule to tie the LED color/status to the status of the relay, it could be possible to make a rule that matches the original firmware where color = a certain power amount

motolav, 2022-09-23:
@crossan007 I don't use tasmota but ESPHome, heres the config I use that changes the LED color according to wattage https://gist.github.com/motolav/2a838a88e151b248e85ccd024d650826
and for reference if anyone wants to create a rule that mimics the original firmware, the colors and wattage are on the bottom

crossan007, 2022-09-23:
> @crossan007 I don't use tasmota but ESPHome, heres the config I use that changes the LED color according to wattage https://gist.github.com/motolav/2a838a88e151b248e85ccd024d650826 and for reference [...]

NICE!   I love ESPHome and Tasmota pretty much equally 🤣  Thanks for the link!