blakadder/templates #1219: [Site issue] Shawader 2Ch Touch Light Switch (WLS028 2)

daniel3514, 2021-11-10:
Please add: 

Installation was done over serial as Tuya version was to new for Tuya Convert to work.

Open the device and locate the 6 pads on the back of the button PCB (not the relay PCB).
The pinout for the 6 pads is as follows (starting from left if pads is in upper right corner):
3.3V
RXD
TXD
GPIO0 (Boot mode pin)
RES12K
GND
Solder cables to the pins and then do as follows:

Connect GND.
Connect RXD to TX on USB to Serial converter.
Connect TXD to RX on USB to Serial converter.
Connect GPIO0 to GND.
Connect RES12K to GND.
Connect 3.3V.
Disconnect RES12K from GND.
(Optional) make backup of original firmware.
Flash tasmota using the USB to Serial converter.
Setup
Enable the template and execute the following command: Backlog ButtonTopic1; SetOption13 1; ButtonTopic2; SetOption13 1; ButtonTopic2; SetOption13 1;

[image]
[image]
[image]

closed 2021-11-12 by blakadder

daniel3514, 2021-11-13:
hello, there is an error in the image above, this is the correct one: 

[image]

blakadder, 2021-11-13:
Then edit the original post with the new image

daniel3514, 2021-11-13:
Excuse me, but where do I upload the image?