blakadder/templates #1279: [Site issue]

octavius08, 2021-12-13:
For support with Tasmota flashing and configuring the device visit [Tasmota Discord](https://discord.gg/Ks2Kzd4).
For support with Tuya-Convert please use their [GitHub Issues](https://github.com/ct-Open-Source/tuya-convert/issues).

Hello Comunity 
I recently bought an AVATTO 3-way switch, as it turned out, they are no longer delivered with an ESP2866 chip. 
And here is my solution after 6 hours of work :)

The current AVATTO 3-way v4.1 switches are no longer with a TYWE3S module but with a WB3S module. 
[image]

However, it is possible to change the WB3S module for an ESP2866 12F. This procedure is time-consuming and need some modifications on the board:
[image]
[image]

The old template no longer works either, here the working template for v4.1 AVATTO 3 Way:
{"NAME":"AVATTO 3 Gang v4.1","GPIO":[0,226,544,34,32,33,0,0,288,224,289,225,290,0],"FLAG":0,"BASE":18}

blakadder, 2021-12-13:
Using ESP-12S or ESP-12H will remove the need for the additional resistor

octavius08, 2021-12-13:
> Using ESP-12S or ESP-12H will remove the need for the additional resistor

Thanks, I didn't know that, I only had the 12F there. 
Finally I solder a SMD 10k resistor directly to the board inbetween GPIO15 and GND

closed 2022-04-06 by blakadder