blakadder/templates #1288: [Site issue]

maceace, 2021-12-22:
For support with Tasmota flashing and configuring the device visit [Tasmota Discord](https://discord.gg/Ks2Kzd4).
For support with Tuya-Convert please use their [GitHub Issues](https://github.com/ct-Open-Source/tuya-convert/issues).

Good day.

I'm not sure I'm writing in the right place.
I bought an aroma diffuser with a TYWE3S module, the same as the Gernems 400 ml Aroma diffuser template (YX-088).
I uploaded to the Tasmot module and started configuring both the device and the Home Assistant.
I found some errors in the HA configuration else if - elif and payload_not_avail:
Subsequently, I am almost convinced that they are not the only ones. I have other problems:
cannot set select.diffuser_color_mode is permanently unknown
can't set the light effect to Random, Wake up or any other
diffuser intensity cannot be set low, medium, high.
the only thing i changed is the topic in mqtt but i changed that everywhere in the scripts and in Rule1.
Can you please advise me what I'm doing wrong?
Please revise the template at:
https://templates.blakadder.com/gernems_YX-088.html

Thank you

blakadder, 2021-12-22:
Do the tuyasend commands work from within Tasmota?

Does Tasmota receive any commands when you try changing values in HA?

maceace, 2021-12-22:
> Do the tuyasend commands work from within Tasmota?
> 
> Does Tasmota receive any commands when you try changing values in HA?

Yes, Tasmota accepts commands, some.
I am able to run a 2 and 4 hour timer
I am not able to change the intensity and light effects for example

blakadder, 2021-12-22:
> > Do the tuyasend commands work from within Tasmota?
> > Does Tasmota receive any commands when you try changing values in HA?
> 
> Yes, Tasmota accepts commands, some. I am able to run a 2 and 4 hour timer I am not able to change the intensity and light effects for example

if you're not able to change them using tuyasend commands your dpid configuration is different from that device

maceace, 2021-12-23:
> > > Fungují příkazy tuyasend z Tasmoty? 
I don't know. I put all the commands in all the instructions in the tasmota console and they were all accepted by tasmoto if you ask about this.

maceace, 2021-12-23:
I'm sorry. I didn't want to close.

maceace, 2021-12-23:
The problem could be that I have the select entity in the state unavailable.

closed 2022-04-06 by blakadder