blakadder/templates #1411: owfeel_EN71 - Led and Relay are inverted - subtle

wipeout666, 2022-05-04:
GPIO00 Relay1 and GPIO13 are inverted. The device will work properly even if inverted, since the LED always have the same state as the relay. HOWEVER, this is not true when wifi is connecting; the LED will flash. Therefore, the template, as posted, will have the valve toggle between open and close with a 1 second period. This could eventually burn the gearbox if wifi is down for a signification amount of time.

[image]

blakadder, 2022-08-11:
corrected

closed 2022-08-11 by blakadder

mrmaximas, 2024-04-22:
@wipeout666 may be you have original tuya fw backup for this valve? thanks!

wipeout666, 2024-04-22:
No I don't, sorry