blakadder/templates #1472: Problem with Sinilink XY-WPCE Template

BlueWings172, 2022-06-27:
Hello

Just flashed my Sinilink XY-WPCE with Tasmota and used [this template](https://templates.blakadder.com/sinilink_XY-WPCE.html) to configure it.

I'm wondering if there is something missing in the template as when I push the power button of the computer case, it doesn't do anything. Also in Tasmota the status of the relay is always Off even when I turn it on and when I shutdown my PC from Windows , it just restarts.

It would be great if it is possible to solve any of these issues from the template.

Thanks

diegargon, 2022-07-17:
Hello.

I purchase the USB powered version
https://templates.blakadder.com/sinilink_XY-WPCL.html

And the template is wrong, the PCE version seems similar perhaps work

The Power SW GPIO is wrong GPIO 10 is the right case in my case.

{"NAME":"XY-WPCL","GPIO":[0,0,320,0,0,224,0,32,160,0,0,0,0,0],"FLAG":0,"BASE":18, "CMND":"SO114 1 | Pulsetime 10 | SwitchMode1 2"}

The Relay must we toggle from off-on-off not lock, mine stayed on long time, and shutdown the computer again, that why i added "Pulsetime 10" (it's my first time i don't known if i doing it right but works)

For check if is powered is Switch1, a binary sensor since the button will go back to off.

In my case, the usb powered version, that binary version is related/detecte with the 'power led' signal and works but not turn off the power led of the case (i not fix that yet)

In HA i create a template for a button that read the binary sensors for show the button state

```
switch:
  - platform: template
    switches:
      camaras_salon:
        unique_id: "CamarasSalon"
        value_template: "{{ is_state('binary_sensor.switch1', 'on') }}"
        availability_template: "{{ not is_state('binary_sensor.switch1', 'unavailable') }}"
        turn_on:
          service: switch.toggle
          target:
            entity_id: switch.pcswitch_cam
        turn_off:
          service: switch.toggle
          target:
            entity_id: switch.pcswitch_cam
        icon_template: >-
          {% if is_state('binary_sensor.switch1', 'on') %}
            mdi:power-plug
          {% else %}
            mdi:power-plug-off
          {% endif %}            
```

BlueWings172, 2022-07-17:
@diegargon 

You are absolutely amazing and a genius. I gave up on this device was going to throw it since I did not backup the original Sinilink firmware so it was completely useless.

I tried your templated and it worked !

My PC no longer restarts when it click shutdown in Windows and the power button works now!
The LED does not seem to be on when the PC is on but for me this does not matter at all.

Also, I've tried your HA template and it also works.

Thank you so much buddy!

closed 2022-07-19 by BlueWings172

dkmode, 2024-02-01:
Thanks for template, works good.

Just a comment: on XY-WPCL USB version, two bottom pairs of power LED connectors are just wired to each other, but on the second pair (POWER LED2) there is a voltage drop from 5V on my mainboard to about 2V, prob. cause of optocoupler and R20 or sth.
This voltage is simply not enough for blue LEDs. Red LEDs works fine (depends on LED color) and I just had to use HDD led for power status. :d

arokoss, 2024-08-26:
[image]

One little Update on this topic for everyone stumbeling on it via search engine: 

I can confirm that the internal electronics of the device result in a voltage drop that causes the case power LED to not work properly. A workaround is to use GPIO04, which is wired to the little button on the board, that is used by the original firmware. 

Steps: 

1. Desolder the button or cut the copper line on the back of the board that leads to to button (there is only one, you cant miss it) 
2. clean the lane that leads to the button (coming from GPIO04 on the chip) on the back of the board by scraping of the paint
3. solder a wire from the open copper to the most inside pin of the led pins (Pin A in the Picture). Now GPIO04 is connected to the + Pin of LED1
4. now you need to cut the lane that connects Pin1+ and Pin2+ on the back of the board, to avoid the direct connection of GPIO12 and GPIO04. 

Thats it - now you can control the LED in the case with GPIO4 and detect the LED output of the Motherboard using GPIO12. 

from the Picture: 
Pin A -> GPIO04
Pin C -> GPIO12
Pin B,D -> Ground

Hope this helps.